Azure/iot-central-firmware #7: Connection string/telemetry data fields not found on device config page

benbp, 2018-05-15:
I have been running through the docs for setting up the MXChip (https://docs.microsoft.com/en-us/azure/iot-central/howto-connect-devkit), and step 7 shows that I should be able to navigate to 192.168.0.1/start in order to set up the connection string and telemetry data. I'm having two issues here:

1. The start page is not found (/start), but it looks like a recent commit moved the HTML into [httpHtmlData.h](https://github.com/Azure/iot-central-firmware/blob/bba79161812b95ebffbb0acb1cb773c53568ba4c/AZ3166/inc/httpHtmlData.h#L14)
2. On the home page, all I am seeing is text fields for WiFi network and password, and no fields for connection string and telemetry data as is detailed in the docs.

Am I doing something wrong here? Are the docs out of date? Would love any assistance. Thanks!

obastemur, 2018-05-15:
@benbp how did you deploy to your board? i.e. you might download the `bin` file from https://github.com/Azure/iot-central-firmware/releases and drag-drop into your board (it should appear as a new external disk drive)

benbp, 2018-05-15:
@obastemur I did the drag and drop method to the USB device. The board is on 1.3.5.53 (which it reported as latest on the display before I upgraded), though it looks like the latest release from your link is 1.2.0. Perhaps I'm too bleeding edge?

obastemur, 2018-05-16:
Sorry for the confusion. `1.3.5` is the bootloader version while `1.2.0` is our sample project version. They are both latest. `1.2.0` is compatible with `1.3.5`. So, if you want to see the demo; go ahead and drag and drop `1.2.0` from our releases.

obastemur, 2018-05-16:
Also added `Compatible with latest MXCHIP boot loader 1.3.5+` line to our release log.

benbp, 2018-05-17:
Sorry @obastemur haven't had a chance to test. I assume it will work but will holler if not. Thanks for the help!

385, 2019-12-05:
Works like a charm! The new firmware was the solution 👍🏽