Azure/iot-central-firmware #13: after coping program.bin int AZ3166 via USB and enter AP can't find /START page

guybartal, 2018-06-17:
I've copied [program.bin](https://github.com/Azure/iot-central-firmware/releases/download/mxchip-v1.3.0/program.bin) into my AZ3166, and after reboot I've entered AP mode, connected with my PC to http://192.168.0.1/start and i got error page with says start page is not found,
are you sure you've burn the image along with the iot central application?
if i go to the root at http://192.168.0.1 i get the standard WiFi config without iot hub connection string and sensors.

I've also confirmed it by configuring IoT Hub connection string via Putty,  i'v got the app with 3 counters: 

> -- Connected --
> sent:xx
> fail:xx
> twin:xx/xx

obastemur, 2018-06-18:
@guybartal just tried from scratch and it worked for me. Can you please send a screen shot of 192.168.0.1 render?

My guess is that.. 192.168.0.X is also what your home / work wifi DHCP provides. Is it possible that your pc was connected to another wifi instead?

obastemur, 2018-06-19:
will close this for now as it's not reproducible. feel free to open it back.

closed 2018-06-19 by obastemur