Azure/iot-central-firmware #22: error connecting to iot central after updating to firmware 2.0.0

guybartal, 2018-10-07:
[image]

obastemur, 2018-10-07:
There are breaking changes to Azure Iot Central device provisioning approaches. Have you reconfigured the device?

obastemur, 2018-10-07:
BTW; a reason for your old connection string is not working might be related to hub changes. New symm key etc. auth approaches dynamically switch the hubs in case the previous one is no longer available.

guybartal, 2018-10-08:
I've noticed those changes, so I've loaded the new iot centra firmware + sample app (v2.0.0) to my MXchip and registered it to IoT Central with all the new configuration IDs. after reboot I got this error...

obastemur, 2018-10-08:
Have you associated the device from web ui? Under devices explorer > unassociated

guybartal, 2018-10-09:
the device was already associated... I've tried to delete it and register it again and it worked.

obastemur, 2018-10-11:
closing this, taking it offline.

closed 2018-10-11 by obastemur