Azure/iot-central-firmware #24: IrDa send information

wgcv, 2018-10-24:
Hi
I'm checking the IRremote library for arduino but doesn't work with the az3166; also I'm checking the documentation (https://microsoft.github.io/azure-iot-developer-kit/docs/apis/irda/) but I don't know how to send a infrared command with *char .

For example in arudino with IRremote is like this
```
irsend.sendNEC(0xFFE896, 32);
```

how i send in AZ?

```
unsigned char powerOn2 = 0xFFE896;
IRDASensor *IrdaSensor;
setup(){
IrdaSensor = new IRDASensor();
IrdaSensor->init();
}
loop() {
int irda_status = IrdaSensor->IRDATransmit(&powerOn, 1, 100 );
if(irda_status != 0)
{
Serial.println("Unable to transmit through IRDA");
}
}
```

But I don't know if the protocol is ok? Because doesn't work.

obastemur, 2018-10-24:
I don't have a receiver at the moment to test. @ArthurMa1978 can you comment on this?

wgcv, 2018-10-25:
For example I'm using LG controller
With the arduino and IRremote i get this:
```
20DF10EF
Decoded NEC: 20DF10EF (32 bits)
Raw (68): 9050 -4450 600 -500 600 -500 600 -1650 600 -550 600 -500 600 -500 600 -550 600 -500 600 -1650 600 -1650 600 -500 600 -1650 600 -1650 600 -1600 650 -1600 600 -1650 650 -450 650 -500 600 -500 600 -1650 600 -500 650 -500 600 -500 650 -500 600 -1600 650 -1600 600 -1650 650 -450 600 -1650 650 -1600 600 -1650 600 -1650 600 
```

But if i use 
```
// Copyright (c) Microsoft. All rights reserved.
// Licensed under the MIT license. 
// To get started please visit https://microsoft.github.io/azure-iot-developer-kit/docs/projects/connect-iot-hub?utm_source=ArduinoExtension&utm_medium=ReleaseNote&utm_campaign=VSCode
#include "AZ3166WiFi.h"
#include "AzureIotHub.h"
#include "DevKitMQTTClient.h"

#include "config.h"
#include "utility.h"
#include "SystemTickCounter.h"

#include "Sensor.h"
IRDASensor *IrdaSensor;
static unsigned char counter;

//////////////////////////////////////////////////////////////////////////////////////////////////////////
// Arduino sketch
void setup()
{
  Screen.init();
  Screen.print(0, "IoT DevKit");

  IrdaSensor = new IRDASensor();
  IrdaSensor->init();
}

void loop()
{
  counter = 0x20DF10EF;

  for (int i = 0; i < 3; i++) {

  int irda_status = IrdaSensor->IRDATransmit(&counter, 32, 100 );
  delay(1000);
}

}
```
The arduino doesn't read any code. I think that is a problem with the encode (IRremote encode with NEC) and IrdaSensor->IRDATransmit only send plain. (Also i don't know what is size and timeout; and pdata is ok like 0x20DF10EF?)

ArthurMa1978, 2018-10-29:
As my answers in the gitter:
The code is not correct. Here the  *counter* is *unsigned char*, so it will be truncated if you assign 0x20DF10EF to this variable. And the second parameter is the size of the buff in byte, nit bit.
So the right code is just like:
```
static unsigned char counter[] = { 0x20, 0xDF, 0x10, 0xEF };

int irda_status = IrdaSensor->IRDATransmit(counter, sizeof(counter), 100);
```

obastemur, 2018-10-29:
@ArthurMa1978 thanks for the help. closing for now

closed 2018-10-29 by obastemur

wgcv, 2018-10-29:
Thanks @ArthurMa1978 !

mchute84, 2019-02-22:
Can I ask a quick question - Did you manager to get the IR signal emitting with the code/changes above?