Azure/iot-central-firmware #25: How to register many devices using x509 certificate.

akubicharm, 2018-10-25:
I'd like to register 20 devices to IoT Central using RaspberryPi firmware samples using x509 certificate.
When I use cert file which is generated by dice_device_enrollment tool, all devices communicate with IoT Central as 'riot-device-cert'.   I'd like to know how to register devices which has individual register id.
Regards,
Kei

obastemur, 2018-10-25:
Azure IoT Python SDK is currently based on C SDK. So, using the demo approach is not really practical. (sample cert is being embedded) This is because you would have to compile the SDK per each device.

If you want to use Python SDK, you can use the Symmetric key instead? Is this an option?

akubicharm, 2018-10-26:
According to "https://github.com/Azure/iot-central-firmware/tree/master/RaspberryPi", now I'm trying to compile SDK using Symmetric key.

The document specify to set SYMMETRIC_KEY_VALUE and REGISTRATION_NAME under hsm_client_key.c, but I don't know how to set that 2 values. 

In addition though I checked out dps_symm_key branch and compiled whole code, but ProvisioningSecurityDeviceType.SAS is not enabled.

I'd like to know how to prepare?

Regards,

Kei

obastemur, 2018-10-29:
@akubicharm I just sent a PR with easier symmkey option -> https://github.com/Azure/iot-central-firmware/pull/27 please let me know if and what part is not clear so I can improve the PR. Thanks!

obastemur, 2019-04-17:
details are available under https://github.com/Azure/dps-certgen/tree/master/samples/python

closed 2019-04-17 by obastemur