Azure/iot-central-firmware #29: MXChip - Http connection failed to connect

theonesuperdave, 2018-12-04:
I have an issue with the MXChip where the device is able to connect to WiFi, but then unable to make an outbound HTTP connection to Azure IoT Central. Looks like it is somewhere in the `iotHubClient.cpp`. This is what I was able to get from the device on Serial Port 9600:
```
Info:    DPS Version: 1.2.8

Info: Iothub Version: 1.1.28

Error: Time:Thu Jan  1 00:41:44 1970 File:prov_transport_http_client.c Func:on_http_connected Line:208 Http connection failed to connect
Error: Time:Thu Jan  1 00:41:45 1970 File:tlsio_mbed_os5.c Func:tlsio_mbedtls_open Line:552 Underlying IO open failed
Error: Time:Thu Jan  1 00:41:45 1970 File:uhttp.c Func:uhttp_client_open Line:1064 opening xio failed
Error: Time:Thu Jan  1 00:41:45 1970 File:prov_transport_http_client.c Func:create_connection Line:602 failed to open http client
Error: Time:Thu Jan  1 00:41:45 1970 File:prov_transport_http_client.c Func:prov_transport_http_open Line:747 Failure creating http connection
Error: Time:Thu Jan  1 00:41:45 1970 File:prov_device_ll_client.c Func:Prov_Device_LL_Register_Device Line:804 Failure establishing  connection
Error: Time:Thu Jan  1 00:41:45 1970 File:DevkitDPSClient.cpp Func:DevkitDPSClientStart Line:302 failed calling Prov_Device_LL_Register_Device
Error: Time:Thu Jan  1 00:41:45 1970 File:DevkitDPSClient.cpp Func:DevkitDPSClientStart Line:320 registration failed!

X - Error at /src/iot-central-firmware/AZ3166/BUILD/sketch/src/iotHubClient.cpp:40
	ERROR: DPS client for GroupSAS has failed.
X - Error at /src/iot-central-firmware/AZ3166/BUILD/sketch/src/iotHubClient.cpp:257
	IoTHubClient::closeIotHubClient!
```

Any ideas where I can get more details or maybe you know of a fix? I am running the latest firmware (2.0.0), and have tried various config options, but it still fails, telling me to reset the device.

The error comes from a corporate network; **it all connects and works fine from my home network.** Please advise at your convenience.

obastemur, 2018-12-04:
@theonesuperdave have you checked the IoT Central UI (device explorer -> unassociated devices) and associate the device id with a template?

obastemur, 2018-12-04:
Just to make sure.. You are no longer seeing that device on `unassociated devices`, right?

I would suspect from DNS is not able to resolve the time servers hence the future expiration time wasn't set properly. Do you have a custom DNS set on your WiFi?

obastemur, 2018-12-04:
I see. Probably it's DNS (time servers). I've got a similar issue with one of the networks at the office. The only fix I know off is to edit the source codes and force set device DNS to gateway IP manually. Perhaps, OS layer could manage this better.

theonesuperdave, 2018-12-04:
Ah, so I either need to get fixed the time of my DNS server to match Azure IoT Central or hack the DNS entries manually. What about simply setting a `host` entry locally to the IP address? This is just a POC.

obastemur, 2018-12-04:
Global time servers not related to Azure IoT Central or Azure etc. (pool.ntp.org)

Do the business network need sign-in etc. steps? looks like, the whole http stack is having problem there. You may connect mxchip to your mobile hotspot?

theonesuperdave, 2018-12-04:
Ok, more or less what I thought. Thanks. And yes, no login page in the middle. I found other `NtpClient` warnings in the event logs for my workstation, so definitely points to the problem there. Will see if I can get those resolved and give it another go.

obastemur, 2018-12-04:
Fingers crossed :) closing this one for now

closed 2018-12-04 by obastemur

theonesuperdave, 2018-12-04:
Just putting this here in case someone else runs into the issue. This turned out to be a bad DNS server still active which was preventing the Azure endpoint from being resolved. Once that was removed, everything works as expected.

My only ask is that the docs be updated to include this plus a few details which point us to serial port watching on port 9600 in VS Code (screenshot below).

[image]