Azure/iot-central-firmware #31: Be more clear on steps and output of start.sh during initial setup with X509

kurtng, 2018-12-13:
I think it should be more clear here what outputs you get during initial setup procedure for X509 certificates.
After ./start.sh first time execution the device is created in IOT Central but the command says DEV_AUTH_ERROR which makes me think something went wrong.
The documentation should state this or the command should be more user friendly in its response and informative.
You should then stop and restart the command when you set a template for the device in IOT Central. Then the second execution of command succeeds.

obastemur, 2018-12-13:
@kurtng  we are aware of the pain points for Python SDK and working on them. I didn't really understand what went wrong with following the steps. Is this something you want to / can contribute or give more details?

kurtng, 2018-12-18:
when i run ./start.sh first time it says DEV_AUTH_ERROR, which made me think that something was wrong. But only think I needed to do was run it once again. The documentation should point that more clearly.

obastemur, 2018-12-18:
`DEV_AUTH_ERROR` appears when a device is not associated to a particular template. (assuming the creds you put there are correct). We are working on improving the messaging and the behavior there. I will keep this here to track that issue.

obastemur, 2019-04-17:
iot central x509 sample is available from https://github.com/Azure/dps-certgen/tree/master/samples/python

closed 2019-04-17 by obastemur