Azure/iot-central-firmware #33: Error when connecting to WiFi (after setup) (AZ3166)

polarapfel, 2018-12-13:
Hi there,

I am exposing a WiFi hotspot on a Windows 10 host. The WiFi is a 2.4Ghz network.

I have successfully flashed the device with the latest firmware, set the access credentials to IoT Central and provided the correct WiFi credentials for a WiFi hotspot exposed from a Windows 10 host.

The AZ3166 shows up with its MAC as connected to the WiFi on the Windows 10 host. However, the device shows "WiFi Connecting..." and then "Error: Please reset the device."

No amount of resetting or hard resetting and running through the setup solves the issue.

Any idea how to fix this?

thanks,

Tobias

obastemur, 2018-12-13:
Logs would help. Try https://microsoft.github.io/azure-iot-developer-kit/docs/use-configuration-mode/ 

Use 9600 as baud rate instead of 115200

obastemur, 2019-01-07:
Closing for now

closed 2019-01-07 by obastemur

sgazinski, 2019-03-31:
Hi there I have exactly the same problem. From the videos and instructions connecting to azure iot hub seemed to be very easy but I am stuck getting the mxchip connected to my wifi. It always says 'WiFi Connecting...'
I have the plan to connect to an iphone (tethering) and in fact the phones says there is one connection but the chip is continiously trying to connect and every 60 seconds the displays shows 'Starting press a and b button or hard reset ...'. Can you give some further idea what to do now

obastemur, 2019-03-31:
@sgazinski have you looked at the logs? See my comment above for how to do it

sgazinski, 2019-03-31:
Sorry but I have difficulties to understand this ... I have not worked with putty yet but I downloaded it yesterday... I have not worked with VS so I have problems in finding out what is the com port ... is there another way to find out which com port is used... maybe then I am able to connect with putty and find the logs

sgazinski, 2019-03-31:
okay i made some progress ...

ERROR: Failed to connect Wi-Fi Ipۛ[;▒s
  - Sensonh▒{cc+▒▒:initSensors
X - Error at /src/MXCHIP/mxchip_advanced/BUILD/sketch/  - WiFiController::initWi                  Fi
ERROR: Failed to connect Wi-Fi Iphonegz.
  - SensorController::initSensors
X - Error at /src/MXCHIP/mxchip_advanced/BUILD/sketch/src/telemetry.cpp:29
        WiFi - NOT CONNECTED - return
▒▒▒▒
▒▒r▒▒▒▒6
        s▒▒+▒▒&▒▒▒{ss#&+▒+▒F{Vs▒+▒6{s3K;▒▒
                                          ▒K{s▒{#+s▒
                                                            2. press Button B and reset to enter AP mode.

  - Configuration found entering telemetry mode.
  - WiFiController::initWiFi
ERROR: Failed to connect Wi-Fi Iphonegz.
  - SensorController::initSensors
X - Error at /src/MXCHIP/mxchip_advanced/BUILD/sketch/src/telemetry.cpp:29
        WiFi - NOT CONNECTED - return

sgazinski, 2019-03-31:
now i get this
 mode.

  - Configuration found entering telemetry mode.
  - WiFiController::initWiFi
Wi-Fi Iphonegz connected.
Time from pool.ntp.org, now is (UTC): Sun Mar 31 18:45:19 2019

  - WiFi WL_CONNECTED
  - SensorController::initSensors
  - AzureIOTClient::init START
  - SCOPEIError: Time:Sun Mar 31 18:45:21 2019 File:C:\Users\v-zuhu\AppData\Local\Arduino15\packag                                                           es\AZ3166\hardware\stm32f4\1.5.1\cores\arduino\azure-iot-sdk-c\c-utility\src\base64.c Func:Base64_                                                           Decoder Line:201 Invalid length Base64 string!
Error: Time:Sun Mar 31 18:45:21 2019 File:C:\Users\v-zuhu\AppData\Local\Arduino15\packages\AZ3166\                                                           hardware\stm32f4\1.5.1\cores\arduino\azure-iot-sdk-c\provisioning_client\src\prov_auth_client.c Fu                                                           nc:sign_sas_data Line:175 Failed decoding symmetrical key
Error: Time:Sun Mar 31 18:45:21 2019 File:C:\Users\v-zuhu\AppData\Local\Arduino15\packages\AZ3166\                                                           hardware\stm32f4\1.5.1\cores\arduino\azure-iot-sdk-c\provisioning_client\src\prov_auth_client.c Fu                                                           nc:prov_auth_construct_sas_token Line:604 Failure generate sas token.
Error: Time:Sun Mar 31 18:45:22 2019 File:C:\Users\v-zuhu\AppData\Local\Arduino15\packages\AZ3166\                                                           hardware\stm32f4\1.5.1\cores\arduino\azure-iot-sdk-c\provisioning_client\src\prov_device_ll_client                                                           .c Func:prov_transport_challenge_callback Line:169 Failure to import the provisioning key
Error: Time:Sun Mar 31 18:45:22 2019 File:C:\Users\v-zuhu\AppData\Local\Arduino15\packages\AZ3166\                                                           hardware\stm32f4\1.5.1\cores\arduino\azure-iot-sdk-c\provisioning_client\src\prov_transport_mqtt_c                                                           ommon.c Func:create_connection Line:578 Failure retrieving sas token from key
D:Error: Time:Sun Mar 31 18:45:22 2019 File:C:\Users\v-zuhu\AppData\Local\Arduino15\packages\AZ316                                                           6\hardware\stm32f4\1.5.1\cores\arduino\azure-iot-sdk-c\provisioning_client\src\prov_transport_mqtt                                                           _common.c Func:prov_transport_common_mqtt_dowork Line:896 unable to create amqp connection
 0nError: Time:Sun Mar 31 18:45:23 2019 File:C:\Users\v-zuhu\AppData\Local\Arduino15\packages\AZ31                                                           66\hardware\stm32f4\1.5.1\cores\arduino\azure-iot-sdk-c\provisioning_client\src\prov_device_ll_cli                                                           ent.c Func:on_transport_registration_data Line:552 Failure retrieving data from the provisioning s                                                           ervice
e000assertion "errorCode == 0" failed: file "/src/MXCHIP/mxchip_advanced/BUILD/sketch/src/AzureIOT                                                           Client.cpp", line 144, function: void AzureIOTClient::init()

obastemur, 2019-03-31:
Glad you passed WiFi connection part. 

The error above says that the device key (primary/secondary..) is not the one you are supposed to. You grab the key from (Device Explorer => click to Device => click to 'Connect' button on top right) ?

sgazinski, 2019-04-01:
Yes but maybe I made a mistake there... In the configuration page there is only one field for the key
What I did ... I wrote  pkey and skey in the field seperated by a '/'. Is this correct...? Or do I have to just put them in one after the other without any seperator...?

obastemur, 2019-04-01:
pick one and use that alone

sgazinski, 2019-04-01:
okay I try that now

sgazinski, 2019-04-01:
without success! I can not retreive the PIN Code... the chip does not come up with this message:
Connect HotSpot:
AZ3166_??????
go-> 192.168.0.1
PIN CODE xxxxx

obastemur, 2019-04-01:
press reset button and then A and B buttons together?

obastemur, 2019-04-01:
feel free to open that as a separate issue. Glad you made it

interfacemirror, 2019-10-28:
@obastemur  I am having a similar problem here. The device is set to use my old WIFI. When I try to reconfigure the device it simply not allowing me to do. I tried using 115200 and 9600 ports. It simply says that it is not able to connect to my wifi. I tried resetting by pressing B and Reset. It only shows to me "NO WIFI" message all the time. I also flushed out the firmware few times with no luck.