Azure/iot-central-firmware #37: sketch/src/config.cpp:119:5: error: 'sasKey' was not declared in this scope

andyt530, 2019-01-06:
Hi,

Working with MXCHIP AZ3166  (v1.6.0)
On macOS with the latest (fixed version) for iotz  (https://github.com/Azure/iotz/issues/51)
Trying to get a customised version of mxchip_advanced for IOT Central:

Defining my own wifi in definitions.h
```
#define WIFI_NAME "my wifi"
#define WIFI_PASSWORD "mypassword"
.....
#define COMPILE_TIME_DEFINITIONS_SET
```
$ iotz compile
sketch/src/config.cpp: In static member function 'static void ConfigController::readGroupSXKeyAndDeviceId(char*, char*, char*, char&)':
sketch/src/config.cpp:119:5: 
**error**: 'sasKey' was not declared in this scope  sasKey = true;  

Looks like that is defined in   onboarding.cpp  ->   OnboardingController::processResultRequest
Which perhaps doesn't get called if you define your own wifi parameters ??

Andy

obastemur, 2019-01-06:
Are you trying to set wifi params only? How about device credentials?

andyt530, 2019-01-06:
Yes the full file is below ......

#define WIFI_NAME "MyName"
#define WIFI_PASSWORD "mypassord"

// #define IOT_CENTRAL_CONNECTION_STRING "put the iot central connect string here"

// use the ones below for SAS or X509
// REQUIRED for SAS
#define IOT_CENTRAL_SAS_KEY "sqIl+vq2............................................"
#define IOT_CENTRAL_REGISTRATION_ID "yyyyyyyyyyy-xxxx-xxxx-xxxx-.........................."

// REQUIRED for both SAS and X509
#define IOT_CENTRAL_SCOPE_ID "0nexxxxxxxxxxxxx"

// REQUIRED for X509
// #define IOT_CENTRAL_USE_X509_SAMPLE_CERT

// #define DISABLE_HUMIDITY
// #define DISABLE_TEMPERATURE
#define DISABLE_SHAKE
#define DISABLE_GYROSCOPE
#define DISABLE_ACCELEROMETER
// #define DISABLE_MAGNETOMETER
// #define DISABLE_PRESSURE

// if you have set any of the above(telemetry) uncomment the one below
#define COMPILE_TIME_DEFINITIONS_SET

obastemur, 2019-01-06:
Looks good. Replace ‘sasKey = true;’ with atype = 'S';

obastemur, 2019-01-06:
Please also comment out ‘#define IOT_CENTRAL_CONNECTION_STRING "https://myconnect.azureiotcentral.com"’

andyt530, 2019-01-06:
Yes - did both and now have new compiled firmware - thank you.

andyt530, 2019-01-06:
Just noticed - anything above a 30 second interval:
#define TELEMETRY_SEND_INTERVAL 30000

And the device goes into a loop - always "Starting ....."   again after 35 seconds - display gets as far as sent: 2  and it restarts every time.

Doesn't happen if TELEMETRY_SEND_INTERVAL is 30000 or less 

Is there some kind of watchdog / timeout I need to change ?

andyt530, 2019-01-06:
I set the timer to longer than that (and it seems to be ok). Either way it's now sending telemetry every 6 minutes without restarting which it what I wanted.

obastemur, 2019-01-07:
Sounds great.

closed 2019-01-07 by obastemur