Azure/iot-central-firmware #39: trouble using new pi python app

goncaloruao, 2019-01-15:
Hi guys !

I'm following the new tutorial from a few hours ago .

But after running the new app on my pi i get the following error:

ERROR : (_mqttConnect) ('ERROR: non JSON is received from %s => %s', ('global.azure-devices-provisioning.net', b'{"operationId":"4.115980cfc72fed8b.6dd9fcff-78d5-4f23-99a7-d84fc34a5263","status":"assigning"}'))

My pi changed from registed to Provisioned inside iot central after running app.py for the first time.

Can anyone point me on the right direction please ? 

Thanks in advance!

obastemur, 2019-01-15:
Python 3.6+ or 2.7+ are the requirements.

obastemur, 2019-01-15:
I did just release an update package and dropped the supported version from 3.7+ to 3.4+.. so it should cover your device without anything else.

```
pip install iotc --upgrade
```

Make sure you have iotc 0.1.6+

Thanks for reporting this issue. closing for now. feel free to open it back if it's not solved for you

closed 2019-01-15 by obastemur

goncaloruao, 2019-01-16:
Thanks for the support! It's already working fine on my device!