Azure/iot-central-firmware #46: fatal error: azure_prov_client/prov_security_factory.h

gregdegruy, 2019-02-05:
Am I missing a dependency, how do I resolve this?

`sketch/src/iotc/sdk_plat/iotc.cpp:14:53: fatal error: azure_prov_client/prov_security_factory.h: No such file or directory`

obastemur, 2019-02-05:
What exactly you are doing?

gregdegruy, 2019-02-05:
Trying to upload this Arduino sample code as is from [here](https://github.com/Azure/iot-central-firmware/tree/master/MXCHIP/mxchip_basic). Only changed the commented sections.

[image]

gregdegruy, 2019-02-07:
@obastemur any feedback on this? Also raised [SO issue](https://stackoverflow.com/questions/54565710/azure-iot-sdk-c-set-up-a-macos-mac-os-x-development-environment) https://stackoverflow.com/questions/54565710/azure-iot-sdk-c-set-up-a-macos-mac-os-x-development-environment

obastemur, 2019-02-07:
@gregdegruy can you share some details please?

Go into board manager and check the mxchip firmware version please

gregdegruy, 2019-02-08:
Pulled down the MXCHIP basic example for [Azure IoT Central](https://github.com/Azure/iot-central-firmware/tree/master/MXCHIP/mxchip_basic) as is and get this error.

sketch/src/iotc/sdk_plat/iotc.cpp:14:53: fatal error: azure_prov_client/prov_security_factory.h: No such file or directory
I'm pretty sure I need to [Set up a macOS (Mac OS X) development environment](https://github.com/Azure/azure-iot-sdk-c/blob/master/doc/devbox_setup.md#macos), but I'm getting stuck at the [Building the C SDK with CMake directly step](http://Building%20the%20C%20SDK%20with%20CMake%20directly). I use cmake to build the SDK and then what? Even after completing this setup I am stilling getting this missing library error.

gregdegruy, 2019-02-08:
Making progress! Switched to the Arduino IDE, installed the latest Azure IoT C libs, and ensured I have the latest MXCHIp AZ3116 board installed in the IDE board manager

Have a new error now.
```c
Build options changed, rebuilding all
sketch/src/iotc/sdk_plat/iotc.cpp: In function 'int prov_dev_set_symmetric_key_info(const char*, const char*)':
sketch/src/iotc/sdk_plat/iotc.cpp:19:78: error: 'hsm_client_set_registration_name_and_key' was not declared in this scope
     hsm_client_set_registration_name_and_key(registration_name, symmetric_key);
                                                                              ^
exit status 1
Error compiling for board MXCHIP AZ3166.
```

[image]

[image]

gregdegruy, 2019-02-08:
I'll raise a new issue for this new error, main issue was fixed by updating the MXCHIP firmware version and installing all of these Azure IoT Libs. Compiles and gives the same output in both VS Code and Arduino IDE.

closed 2019-02-08 by gregdegruy