Azure/iot-central-firmware #47: Adding State information in Raspberry PI sample

rg6512, 2019-02-08:
I modified my device template to include a Temperature State of Normal, Warning, or Critical.  I modified the code to evaluate the temperature and establish the state.  The state messages are being sent to IoT Central but the information is not available for display.  The simulated device in IoT Central displays the state correctly.  Any help would be appreciated.
Here is the section of the code:

while iotc.isConnected():
  iotc.doNext() # do the async work needed to be done for MQTT
  if gCanSend == True:
    if gCounter % 10 == 0:
      gCounter = 0
      temp = randint(65, 85)
      if temp > 29:
         tempStatus = "Critical"
      elif temp > 27:
         tempStatus = "Warning"
      else:
         tempStatus = "Normal"
      iotc.sendState("{ \"tempStatus\": " + tempStatus + " }")

obastemur, 2019-02-08:
Make sure the status information is upper case and inside the double quotes.

rg6512, 2019-02-08:
The code example I showed above is from a modified Python example  The formatting didn't transfer well when I pasted into the comments.  The code is running and I see the expected callback information.  The telemetry is available but not the state information.  See callback results:

- [onmessagesent] => { "tempStatus": NORMAL }
- [onmessagesent] => { "temp": 22, "humidity": 59, "distance": 42}

obastemur, 2019-02-08:
When you put `NORMAL` without quotes, that wouldn't be a valid JSON since there is no type / value for `NORMAL`. That's why, it should be `"NORMAL"` -> `{ "tempStatus": "NORMAL" }`

rg6512, 2019-02-08:
Thanks.  That worked.  I appreciate the help and insight.

obastemur, 2019-02-08:
Cool, glad to hear that!

closed 2019-02-08 by obastemur