Azure/iot-central-firmware #49: 8266 DPS endpoint PUT call has failed.

SMARTHON, 2019-02-09:
Hi ALL, i got the wifi connected. I am using nodemcu, entered the right information.

However, the below error I got, I double checked the scopeId, deviceId, deviceKey already. Are there any problem on it? Thanks!

⸮Hp⸮⸮>dh⸮⸮xI<⸮  - Connecting WiFi..
- iotc.dps : getting auth...
- iotc.dps : getting operation id...
ERROR: DPS endpoint PUT call has failed.
X - Error at C:\Users\Ken Lai\Downloads\iot-central-firmware-master\iot-central-firmware-master\ESP8266\esp8266_arduino\esp8266_arduino.ino:70
	Error @ iotc_connect. Code 1

Regards,
Ken

obastemur, 2019-02-09:
That error means that it could not reach at the server at all. Possibly a connection issue

SMARTHON, 2019-02-09:
Thanks for your reply! Does it mean the problem on server side? Or the firmware problem? Because i just noticed that the firmware development for 8266 to azure iot central is quite new. Just wonder... thanks!

obastemur, 2019-02-09:
Neither. Most probably the wifi connection.

SMARTHON, 2019-02-10:
Thanks! However, I have just changed another hotspot (different ssid and pw), the same error happened... Do you have any idea on it? Many thanks.

obastemur, 2019-02-11:
Can you share the steps you follow ? and which exact board?

SMARTHON, 2019-02-14:
Dear SIr,

Sorry for the late reply! The board I am using is nodemcu dev kit v1.0 board.

1. For the platform, I follow the steps provided strictly.

- Visit AzureIoTCentral and create a new application.
- Select Sample Devkits
- Add a new mxchip device. (real device) (Under the Device Explorer)
- Browse into device window and click/open Connect on the top-right of the device screen

2. For the firmware, I am using arduino IDE 1.8.X. I download the whole github iot central source, and open the esp8266_arduino.ino with the suitable header/library link changed in the ino. After that, I compiled and uploaded to the board "NodeMCU 1.0 (ESP12E-module)" successfully. Of course, I have done the following.

- Grab scopeId, device Id and primary key and fill the necessary parts under ESP8266.ino
// #define WIFI_SSID "<ENTER WIFI SSID HERE>"
// #define WIFI_PASSWORD "<ENTER WIFI PASSWORD HERE>"
// const char* scopeId = "<ENTER SCOPE ID HERE>";
// const char* deviceId = "<ENTER DEVICE ID HERE>";
// const char* deviceKey = "<ENTER DEVICE primary/secondary KEY HERE>";

However, the error shown. "* ERROR: DPS endpoint PUT call has failed."

Regards,
Ken

obastemur, 2019-02-19:
@SMARTHON "ERROR: DPS endpoint PUT call has failed" happens only when the board is not able to reach at Azure.

The issue could be a connection issue OR a certificate validation issue. Do you know if your router is behind a proxy?

SMARTHON, 2019-02-20:
> @SMARTHON "ERROR: DPS endpoint PUT call has failed" happens only when the board is not able to reach at Azure.
> 
> The issue could be a connection issue OR a certificate validation issue. Do you know if your router is behind a proxy?

I see, I don't think so. 

Because I use normal wifi function in arduino, and I can upload the data to ThingSpeak platform using API. It proves that there is no problem on router. 

Also, I tried the hotspot from my phone, still doesn't work..

Thanks a lot for your help.

obastemur, 2019-02-22:
I’m curious if retry support would fix the issue for you. Are tou able to share all the logs from serial connection?

SMARTHON, 2019-02-22:
Hi sir,

Thanks for your reply.

The whole serial log are below.

====================================================
Connecting WiFi..
- iotc.dps : getting auth...
- iotc.dps : getting operation id...
ERROR: DPS endpoint PUT call has failed.
X - Error at C:\Users\Ken Lai\Downloads\iot-central-firmware-master\iot-central-firmware-master\ESP8266\esp8266_arduino\esp8266_arduino.ino:70
	Error @ iotc_connect. Code 1
====================================================

If you don't mind take some time, i have recorded the video in you youtube. There are whole process I have done.

https://www.youtube.com/watch?v=_phDOBtLw6c&feature=youtu.be

Regards,
Ken

obastemur, 2019-02-28:
@SMARTHON  are you able to follow the steps from README? I made few changes to code (i.e. removed FlasStringHelper since it might not work with some of the boards.. Can you please update the code and follow the steps and let me know the result?

obastemur, 2019-03-01:
possibly you were the victim of TLS lib update on esp8266 headers. Please update to latest code and no matter what version you have, this problem should be fixed.

Busch, 2019-03-02:
I got the exact same issue using the latest code. ESP8266 header firmware is on version 2.5.0 (also doesnt work with versions 2.4.2, 2.4.1 and 2.4.0):

  - Connecting WiFi..
  - - iotc.dps : getting auth...
  - - iotc.dps : getting operation id...
  - ERROR: DPS endpoint PUT call has failed.
X - Error at C:\Users\user\Desktop\ARDESP\ESP8266\ESP8266.ino:70
	Error @ iotc_connect. Code 1

obastemur, 2019-03-02:
You mean the current latest sample doesn’t work for you? Can you share more details please?

obastemur, 2019-03-02:
A bit more info; esp8266 firmware has deprecated axtls and the prev. CA cert interface. However it’s more than that since the new API doesn’t work the way it does before. ie. You need make an extra api call. As a result, we have compile time check in code to determine that and use fingerprint instead of CA cert. 

In case there is an alternative there and neither of these work for you, please share more detail and feel free to contribute

Busch, 2019-03-02:
Thanks for your fast reply. Sure, here's more information about my environment and steps to reproduce:

- Hardware: Node MCU ESP8266
- Arduino Settings:

* Added the additional board manager URL (File->Preferences): http://arduino.esp8266.com/stable/package_esp8266com_index.json
* Installed the Board "esp8266" Module (Tools->Board Manager) in latest version 2.5.0
* Selected "Node MCU 1.0 ESP-12E Module (Tools->Board)

- IoT-Central Firmware:

* Cloned latest https://github.com/Azure/iot-central-firmware to local filesystem
* Edited ESP8266.ino:

#define WIFI_SSID "MYWIFINETWORK"
#define WIFI_PASSWORD "MYWIFIPASSPHRASE"

const char* scopeId = "MYSCOPEID";
const char* deviceId = "MYDEVICEID";
const char* deviceKey = "MYPRIMARYDEVICEKEY";

* Connected the Hardware via COM3
* Pressed Upload:

Sketch uses 411620 bytes (39%) of program storage space. Maximum is 1044464 bytes.
Global variables use 34700 bytes (42%) of dynamic memory, leaving 47220 bytes for local variables. Maximum is 81920 bytes.
Uploading 415760 bytes from C:\Users\user\AppData\Local\Temp\arduino_build_424876/ESP8266.ino.bin to flash at 0x00000000
................................................................................ [ 19% ]
................................................................................ [ 39% ]
................................................................................ [ 58% ]
................................................................................ [ 78% ]
................................................................................ [ 98% ]
.......                                                                          [ 100% ]
  
* Opening Serial Monitor:

 - Connecting WiFi..
  - - iotc.dps : getting auth...
  - - iotc.dps : getting operation id...
  - ERROR: DPS endpoint PUT call has failed.
X - Error at C:\Users\user\Desktop\ARDESP\ESP8266\ESP8266.ino:70
	Error @ iotc_connect. Code 1

How can I further troubleshoot?

obastemur, 2019-03-02:
Can you insert ‘IOTC_LOG(“fingerprint!!!”);’ right before the line 
https://github.com/Azure/iot-central-firmware/blob/master/ESP8266/src/iotc/arduino/iotc.cpp#L86 ? And share the serial logs?

Busch, 2019-03-02:
Sure. Here's the serial log:

  - Connecting WiFi..
  - - iotc.dps : getting auth...
  - - iotc.dps : getting operation id...
  - fingerprint!!!
  - ERROR: DPS endpoint PUT call has failed.
X - Error at C:\Users\user\Desktop\ARDESP\ESP8266\ESP8266.ino:71
	Error @ iotc_connect. Code 1

SMARTHON, 2019-03-02:
@obastemur 
Hi, I updated my code to the latest code from github too. And I also used the latest version for 8266(2.5.0). I still got the same issue with same log..

⸮H⸮⸮h⸮$⸮yI⸮⸮  - Connecting WiFi..
  - - iotc.dps : getting auth...
  - - iotc.dps : getting operation id...
  - ERROR: DPS endpoint PUT call has failed.
X - Error at C:\Users\Ken Lai\Desktop\iot-central-firmware-master\iot-central-firmware-master\ESP8266\ESP8266.ino:70
	Error @ iotc_connect. Code 1

SMARTHON, 2019-03-02:
Hi ALL, For my pc, I have also added the line, below is the same result.

⸮EO|t`MM⸮4C8⸮⸮  - Connecting WiFi..
  - - iotc.dps : getting auth...
  - - iotc.dps : getting operation id...
  - fingerprint!!!
  - ERROR: DPS endpoint PUT call has failed.
X - Error at C:\Users\Ken Lai\Desktop\iot-central-firmware-master\iot-central-firmware-master\ESP8266\ESP8266.ino:70
	Error @ iotc_connect. Code 1

obastemur, 2019-03-02:
Can you please select 2.4.2 for the esp8266 version from board manager and see the result + output ? Thanks!

SMARTHON, 2019-03-02:
Dear ALL, 

1. it works for 2.4.2! (For more info, Mine is arduino IDE 1.8.8 window store 1.8.19.0)

⸮DOz4`LME⸮<C8⸮⸮  - Connecting WiFi..
  - - iotc.dps : getting auth...
  - - iotc.dps : getting operation id...
  - - iotc.dps : getting host name...
  - 
hostname: XXX
deviceId: XXX
username: XXX
password: XXX

  - Is connected ? YES (64)
  - - [ConnectionStatus] event was received. Payload => EMPTY
  - - [MessageSent] event was received. Payload => {"accelerometerX": 23}
  - - [MessageSent] event was received. Payload => {"dieNumber":4}
  - - [MessageSent] event was received. Payload => {"accelerometerX": 12}
  - - [MessageSent] event was received. Payload => {"dieNumber":5}
  - - [MessageSent] event was received. Payload => {"accelerometerX": 10}
  - - [MessageSent] event was received. Payload => {"dieNumber":4}

2. For 2.5.0, I can compile and upload and got below log.

⸮H�⸮⸮h⸮$⸮�yI�⸮⸮ - Connecting WiFi..

iotc.dps : getting auth...
iotc.dps : getting operation id...
ERROR: DPS endpoint PUT call has failed.
X - Error at C:\Users\Ken Lai\Desktop\iot-central-firmware-master\iot-central-firmware-master\ESP8266\ESP8266.ino:70
Error @ iotc_connect. Code 1

3. I have also tried different version before 2.4.2 (They are 2.4,1, 2.4.0, 2.3.0, 2.2.0), All got the below compilation error. I think it is normal.

Build options changed, rebuilding all
In file included from sketch\src\iotc\arduino\comms.cpp:4:0:

sketch\src\iotc\arduino\../common/iotc_platform.h:54:26: fatal error: avr/pgmspace.h: No such file or directory

 #include <avr/pgmspace.h>

                          ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
Could not find boards.txt in C:\Users\Ken Lai\Documents\Arduino\hardware\espressif\esp32. Is it pre-1.5?

SMARTHON, 2019-03-02:
@obastemur Thanks

Busch, 2019-03-02:
I can confirm it is working with version 2.4.2!

Thanks @obastemur and @SMARTHON

obastemur, 2019-03-02:
Cool! I’m happy that you got it working now :) please keep 2.4.2! I’ll be watching out for the developments and investigate this further.

SMARTHON, 2019-03-02:
Thanks @obastemur 
Thank you too! @Busch

closed 2019-03-18 by obastemur

paluigi, 2020-11-04:
Hello, the status is still the same? 2.4.2 is the last working version for ESP8266 board manager?