Azure/iot-central-firmware #51: trying to build Base OS - missing parson.h file

stevebus, 2019-02-13:
fresh install of Ubuntu 16.04 running fresh install of iotz

Trying to build the MXChip base OS, so I can bypass the certificate validation to connect MXChip to IoT Edge.  I'm following the instructions here ->  https://github.com/Azure/iot-central-firmware/tree/master/MXCHIP/mxchip_advanced/build_firmware

When I do the iotz make, I get the following error about a missing parson.h file.  Where does that come from and how do I get it?

stevebus@azureEdgeUbuntu1:~/mxchip/iot-central-firmware/MXCHIP/mxchip_advanced/build_firmware$ sudo iotz make
SDK_DIR=`ls /root/.arduino15/packages/AZ3166/hardware/stm32f4/ | awk '{print }'` && \
SDK_DIR="/root/.arduino15/packages/AZ3166/hardware/stm32f4/$SDK_DIR" && \
cp mxchip_tweak ../.iotz.mxchip.tweak && \
cd .. && \
rm -rf BUILD/build.options.json && \
rm -f $SDK_DIR/system/sdk/lib/libaz_iot.a && \
rm -rf $SDK_DIR/cores/arduino/az_iot && \
rm -rf $SDK_DIR/libraries/AzureIoT && \
ln -s `pwd`/build_firmware/devkit-sdk/AZ3166/src/libraries/AzureIoT $SDK_DIR/libraries/AzureIoT && \
ln -s `pwd`/build_firmware/devkit-sdk/AZ3166/src/cores/arduino/az_iot $SDK_DIR/cores/arduino/az_iot && \
cp `pwd`/build_firmware/devkit-sdk/AZ3166/src/cores/arduino/EEPROMInterface.h $SDK_DIR/cores/arduino/ && \
cp .iotz.mxchip.tweak $SDK_DIR/platform.txt && \
arduino --board 'AZ3166:stm32f4:MXCHIP_AZ3166' --verify 'iotCentral.ino' --pref build.path=BUILD && \
cp $SDK_DIR/bootloader/boot.bin /tools && \
mv BUILD/iotCentral.ino.bin BUILD/iotCentral.ino_no_bootloader.bin && \
python /tools/arduino.az3166.boot_patch.py BUILD/iotCentral.ino_no_bootloader.bin BUILD/iotCentral.ino.bin && echo "DONE!"
Picked up JAVA_TOOL_OPTIONS:
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
**/root/.arduino15/packages/AZ3166/hardware/stm32f4/1.6.1/libraries/AzureIoT/src/DevKitOTAUtils.cpp:2:20: fatal error: parson.h: No such file or directory
 #include "parson.h"**
                    ^
compilation terminated.
exit status 1
Makefile:7: recipe for target 'all' failed
make: *** [all] Error 1

obastemur, 2019-02-14:
There are breaking changes to devkit-sdk build configuration. I'm planning to use the approach from https://gist.github.com/obastemur/f5c7132e1233c7cf2e4c6dce5f548b26 and update the sample.

stevebus, 2019-02-14:
I don't completely follow what that comment means and/or what it buys me in the scenario where I need to make changes to the underlying dev kit and use those in your iotc firmware...

obastemur, 2019-02-14:
The link I shared is a way to compile a very basic mxchip app with the devkit. The trick is to  remove  the main logic file there and put the mxchip_advanced folder instead. However I know it’s not trivial.

This repo doesn’t have any official duty to explain how to compile devkit-sdk and the build script for that purpose is kept seperated under a sub folder. However, I understand the pain point and that’s why the script. I will try to update the script again. Unfortunately, underlying changes to devkit sdk might break the defacto script here again. Appreciate the bug report

stevebus, 2019-02-14:
Ok, got it.. Appreciate the advice and the link.    Been knocking all the rust off of my C skills and get lost sometimes in all the build configs.

obastemur, 2019-02-14:
@stevebus could you please try and verify https://github.com/obastemur/mxchip_az3166_firmware ?

stevebus, 2019-02-14:
yeah - that build process worked great...  going to muck around in the code to see what changes I need to make now, but the build process was very smooth.  thanks!

obastemur, 2019-02-14:
happy to hear.

closed 2019-02-14 by obastemur