Azure/iot-central-firmware #54: ESP32 Compile Error

10c0, 2019-02-18:
Getting the following error when running make flash on ESP32. Tried on two different computers and build stops at same place.

```
AR build/lwip/liblwip.a
make[1]: *** No rule to make target 'iotc/iotc.o', needed by 'libmain.a'.  Stop.
/home/kx5i/esp/esp-idf/make/project.mk:468: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2
```
Only the iotc.h files is copied to the main/iotc directory. I copied the other folders and files there from ~/esp/iot-central-firmware/MXCHIP/mxchip_basic/src/iotc/ to main/iotc and it did not make a difference.

obastemur, 2019-02-18:
What are you trying to achieve?

10c0, 2019-02-18:
https://github.com/Azure/iot-central-firmware/blob/master/ESP32/esp-azure/Makefile does a file copy of only the iotc.h file. So I opted to bring in the rest to see if it made a difference, it did not.

```
PROJECT_NAME := azure_iot_central

ifeq ($(OS),Windows_NT)
$(shell copy ..\..\MXCHIP\mxchip_basic\src\iotc\*.* main\iotc)
else
$(shell cp -f ../../MXCHIP/mxchip_basic/src/iotc/*.* main/iotc)
endif

```

obastemur, 2019-02-18:
Could you please `git pull && git checkout esp32_full` and see if that will work for you?

10c0, 2019-02-19:
The compile completed and flashed. I'll run make monitor next to see what is going on. So far the device is not showing up in the IoTC console.

obastemur, 2019-02-19:
Double check the “unassociated devices”

obastemur, 2019-02-19:
I will close this one since the original `compile error` is no longer the issue. Feel free to write here if you got other problems. Thanks for opening this!

closed 2019-02-19 by obastemur