Azure/iot-central-firmware #56: iotc.isConnected() never fails.

BigBadPhysicist, 2019-02-21:
Hi,

i have iotc running on a Raspberry Pi 3 B+.

I run into the issue, that if the Raspberry Pi is loosing Internet connection, the loop 

while iotc.isConnected() 

never sees that the connection was lost.

The programm keeps running and gCanSend is always true.

I tested this with an LTE Stick running the code and sending telemetry to IoT Central.
As soon as i pull the LTE Stick the mqtt messages timeout but the loop keeps on running and never fails.

obastemur, 2019-02-22:
AFAIK Hub’s mqtt broker supports ping. Thanks for opening this issue, I will take a look at it

BigBadPhysicist, 2019-03-08:
Hi,

I just wanted to know if there is an update?

I can send you additional logs and give you acces to a smalles working example if you need support.

FuzzyPones, 2019-03-08:
Alternatively, MQTT supports Quality of Service. If QoS is 1 the broker should reply with a PUBACK. 
I see that paho mqtt is used and in the doc QoS can be set:
> publish(topic, payload=None, qos=0, retain=False)

Also it returns a MQTTMessageInfo that should have an error if connection fails.

Also, the connect already has an automatic PING implemented:

> keepalive
> maximum period in seconds allowed between communications with the broker. If no other messages are being exchanged, this controls the rate at which the client will send ping messages to the broker

The iotc.py code is closed source or can I find that somewhere?

obastemur, 2019-03-12:
source code is available at https://github.com/obastemur/iot_client/tree/master/python appreciate the contributions!

BigBadPhysicist, 2019-03-28:
Thank you for providing the link.

I looked into your code. 

The problem seems to be the function

```
  def isConnected(self):
    return self._mqttConnected
```

_mqttConnected is set to true as soon as you call _on_connect and is only set to false if you call disconnect.

I did not see any other way _mqttConnected is changed.

This should be included in 

```
  def doNext(self):
    if not self.isConnected():
      return
    if mqtt == None:
      try: # try non-blocking
        self._mqtts.check_msg()
        time.sleep(1)
      except: # non-blocking wasn't implemented
        self._mqtts.wait_msg()
    else: #paho
 # catch error and set mqtt to disconnected 
time.sleep(1)
```
or

```
  def _sendCommon(self, topic, data):
    if mqtt != None:
      (result, msg_id) = self._mqtts.publish(topic, data)
      if result != mqtt.MQTT_ERR_SUCCESS:
        LOG_IOTC("ERROR: (sendTelemetry) failed to send. MQTT client return value: " + str(result) + "")
      # set mqtt to disconnected
        return 1
      self._messages[str(msg_id)] = data
    else:
      self._mqtts.publish(topic, data)
      msg_id = 0
      self._messages[str(msg_id)] = data
      self._on_publish(None, topic, msg_id)

return 0
```

I am still not sure why the connection doesn't recover on its own if one message is lost.

obastemur, 2019-04-29:
0.3.2 fixes it

closed 2019-04-29 by obastemur