Azure/iot-central-firmware #58: Where to add delay for telemetry sending

juhaluoto, 2019-03-04:
I am a bit rookie here, and I have been trying to add delay to the telemetry sending, where would it make sense to add? Post processing of the data is getting difficult, if the telemetry is send so often.

Thank you!

obastemur, 2019-03-04:
@juhaluoto which example? or platform?

juhaluoto, 2019-03-04:
I have been working with the MXChip AZ3166 Devkit using the mxchip_advanced.

obastemur, 2019-03-05:
You could increase the `TELEMETRY_SEND_INTERVAL` value from https://github.com/Azure/iot-central-firmware/blob/master/MXCHIP/mxchip_advanced/inc/globals.h#L37 and that would affect 
https://github.com/Azure/iot-central-firmware/blob/master/MXCHIP/mxchip_advanced/src/telemetry.cpp#L131

juhaluoto, 2019-03-05:
Thank you, I totally missed that. Now I got it and it seems I had to change the watchdog timer to be longer, as I wanted to have telemetry sent once a minute. Many thanks for fast reply!!

closed 2019-03-05 by obastemur