Azure/iot-central-firmware #64: Cannot guarantee that the connection is can be successfully every time

SMARTHON, 2019-04-09:
Hi ALL,

I can connect the azure iot central from 8266 successfully.

However, it doesn't guarantee that it can connect successfully everytime.

The error as show below. Sometime, i try many times as mid-night. It doesn't work. However, i try one time an another time, it can connect right the way. Sometimes, I try several times to make it work. Is it the problem from server side? 

 - - iotc.dps : getting auth...
  - ERROR: couldn't fetch the time from NTP.
  - - iotc.dps : getting operation id...
  - ERROR: DPS (PUT) request has failed.
{"errorCode":401002,"trackingId":"e7c63d1e-5831-4b88-8882-c4a170670f8c","message":"Unauthorized","timestampUtc":"2019-04-09T05:06:23.4508912Z"}
0

X - Error at C:\Users\Ken Lai\Documents\Github_smart\multiple-sensors-plant-microsoft\ESP8266\ESP8266.ino:221
	Error @ iotc_connect. Code 1

obastemur, 2019-04-09:
@SMARTHON from your logs `ERROR: couldn't fetch the time from NTP.` 

The WiFi interface you got does have a problem with resolving the NTP. You may update the code and use a static ip instead of uri if you are unable to fix the DNS on your router

SMARTHON, 2019-04-09:
@obastemur   Thanks for your reply again!

However, I don't think it is the problem from the DNS of the router. The reason is I use the same router, it cannot connect to the azure iot central with the percentage around 30%. It can connect to the azure iot central with the percentage around 70%. 

Sometime, I cannot connect with many trial within 10 minutes. And 10 mins after a while, it can connect to the azure central again. A little bit strange on it.

For the updating code, would you mind to share how to change it a bit more detail? thanks you osastemur.

Regards,
Ken

obastemur, 2019-04-09:
@SMARTHON all the symptoms sound similar. Presumably, the reason is the DNS settings.

obastemur, 2019-04-09:
In the sample, you may replace NTP fetching code to some static future time value.

obastemur, 2019-04-12:
Closing for now. Feel free to open it back.

closed 2019-04-12 by obastemur

paluigi, 2020-11-04:
Hello @obastemur , I have the same issue the OP had. Can you describe how to replace the NTP fetching code? Thanks!