Azure/iot-central-firmware #67: Disconnects after 6 hours due to "not authorized"

SkylarLK9998, 2019-04-20:
Device: Raspberry Pi 3 Model B+
Operating System: Raspbian 9
Python Version: 2.7.13

After running for 6 hours I get: 'on(disconnect): Not authorized'
This is accompanied by 12 or 24 'ERROR: (sendTelementry) failed to send. MQTT client return value: 4' when I am rapid firing messages, but when I'm firing messages at a decreased speed I do not get these messages. In the end the command line creates a new pi@raspberrypi ~ $.

Is anyone else having a problem with losing authorization after 6 hours?

I have tried adding an if loop to the script that will reconnect the device if it disconnects but it didn't work.

obastemur, 2019-04-20:
Are you using a free application or paid one?

SkylarLK9998, 2019-04-23:
Paid application. S1 tier.

SkylarLK9998, 2019-04-23:
This error occurs regardless of if I rapid fire messages or do fire them at a slower pace. I only have 2 devices (both RasPis) on this application and they both crash after 6 hours even when have yet to hit the 400k daily message limit. My Python Script has a counter that keeps track of my message count so know exactly how many messages get sent. The message limit is not the problem.

obastemur, 2019-04-23:
6 hours corresponds to token expiration limit.

SkylarLK9998, 2019-04-26:
I tried an if-statement that calls iotc.connect once the authorization expires and it didn't work. Any ideas?

obastemur, 2019-04-27:
can you please share a sample code for what you did? with some log?

obastemur, 2019-04-29:
you may try the version 0.3.2 which may help you to manage token expiration time and `isConnected` status

changelog
```
- change isConnected status to false if mqtt rc == 1
- new `setTokenExpiration` interface for manually defining token expiration time
```

simply `pip install iotc --upgrade`

closed 2019-05-08 by obastemur