Azure/iot-central-firmware #70: Cannot build using vscode & platform.io

anderl80, 2019-04-29:
I cloned the repo, used eps32 src and put it in this folder structure:
[image]

Tried to build:
[image]

obastemur, 2019-04-29:
looks like a missing header file issue. have you cloned the target repository `recursively`? and have those folders?

anderl80, 2019-04-29:
I downloaded the repo zip and copied the esp32 folder with all subfolders and files.

obastemur, 2019-04-29:
hmm, are you also able to confirm `azure_c_shared_utility/gballoc.h` in path?

anderl80, 2019-04-29:
[image]

Files seem to be missing. `azure_c_shared_utility` is not part of the repository.

anderl80, 2019-04-29:
> 
> 
> hmm, are you also able to confirm `azure_c_shared_utility/gballoc.h` in path?

No cannot find the file.

zhaodong2013062, 2019-05-07:
I just tried the sample on on my machine without platform.io and it works, so I don't think that it is an issue with misplaced files or includes. I'm not familiar with platform.io - perhaps you need to tweak the file structure?

anderl80, 2019-05-07:
ok, thanks

closed 2019-05-09 by obastemur