Azure/iot-central-firmware #72: ESP8266 Settings is not picked up directly after restart of device

sandervandevelde, 2019-05-06:
I sent setting values to my ESP8266 with the latest sketch. 

this was done with a device which was running and connected to the Arduino DE but not connected to the WIFI.

Afterward, when deploying the same sketch with update WIFI settings, the settings are still not sent. The Update button is grayed out and the labels are not showing a new latest update time.

I expect the ESP to check for settings once right after successfully connecting to IoT Central with correct WIFI settings.

obastemur, 2019-05-06:
@sandervandevelde you were possibly using the esp8266 sample code with your own device template? can you be more specific? and make sure you have the latest code.

sandervandevelde, 2019-05-07:
Yes, you are right in both cases. This is about the ESP8266.ino. And I was not using the latest code.

But when I checked the if the same issue still occurred (disconnecting my device, updating the two settings, connecting my device and checking the Serial monitor in the Arduino IDE for the stack trace) I ran into two issues:

ISSUE 1. the device crashed on startup

Yes, the settings are now read on startup. But right after that, I see my device crashing. I get this output in the monitor: (added as file)

[crash on startup.txt](https://github.com/Azure/iot-central-firmware/files/3154303/crash.on.startup.txt)

This is what I checked:

2 settings, integer and bool -> crash after reset of device
1 setting, integer -> success! We get the right desired and reported property and device starts sending telemetry
2 settings, integer and integer -> crash after reset of device
1 setting, integer -> success again! We get the right desired and reported property and device starts sending telemetry

So it seems having two (more than one) properties causes the crash.

ISSUE 2. IoT Central reports strange value when updating the toggle setting

See:

[image]

As you can see, the integer value is change and the old value is shown in yellow. But for the boolean value, the new value is shown instead of the old value.

Conclusion: 

the settings are now picked up but the device crashes constantly when receiving more than one updated setting.

obastemur, 2019-05-07:
You've got a stackoverflow while printing out the payload for settingsUpdated.

Can you share your code? I will try to see if I can repro.

sandervandevelde, 2019-05-07:
Well actually, this is just the standard library, extended with a few extra lines of code:
https://gist.github.com/sandervandevelde/ad55c28fc31c7ea718f6af8d155c1055

obastemur, 2019-05-08:
Not sure how fast you trigger the events. If that's the case, you might need to alloc the payload memory inside the callback handler and re-use. Just checked the base sample and it works fine.

i.e. you updated the sample to send reported property as well. Which will end-up generating callback. Once the callback is created, the logic there will allocate a memory to show the payload. You could show the payload by `......... %.*s", lenght, stringPtr);` and don't allocate from heap etc..

It's a limited device.

closed 2019-05-08 by obastemur

sandervandevelde, 2019-05-08:
Hello, thanks for this information. I will dive into this suggestion. I cannot deny that the device is limited but I was hoping it should be able to receive two settings ;-)

Did you look at the second issue with the IoT Central dashboard? You should be able to see this one too. Can you address this to the IoT Central team?

obastemur, 2019-05-08:
It’s not a central side issue. You got stack overflow before ack is complete. Thus, central can’t confirm that you received the setting update