Azure/iot-central-firmware #74: Connection lost after sending telemetry

anderl80, 2019-05-07:
Somehow the connection is lost after sending a telegram.

```c++
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include "iotc/common/string_buffer.h"
#include "iotc/iotc.h"

#define WIFI_SSID "..."
#define WIFI_PASSWORD "..."

const char *SCOPE_ID = "...";
const char *DEVICE_ID = "...";
const char *DEVICE_KEY = "...";

int pin = D2;
volatile unsigned int pulse;
const int pulses_per_kwh = 1000;
const int interval = 30000;

void on_event(IOTContext ctx, IOTCallbackInfo *callbackInfo);
#include "connection.h"

void on_event(IOTContext ctx, IOTCallbackInfo *callbackInfo)
{
    // ConnectionStatus
    if (strcmp(callbackInfo->eventName, "ConnectionStatus") == 0)
    {
        LOG_VERBOSE("Is connected ? %s (%d)",
                    callbackInfo->statusCode == IOTC_CONNECTION_OK ? "YES" : "NO",
                    callbackInfo->statusCode);
        isConnected = callbackInfo->statusCode == IOTC_CONNECTION_OK;
        return;
    }

    // payload buffer doesn't have a null ending.
    // add null ending in another buffer before print
    AzureIOT::StringBuffer buffer;
    if (callbackInfo->payloadLength > 0)
    {
        buffer.initialize(callbackInfo->payload, callbackInfo->payloadLength);
    }

    LOG_VERBOSE("- [%s] event was received. Payload => %s\n",
                callbackInfo->eventName, buffer.getLength() ? *buffer : "EMPTY");

    if (strcmp(callbackInfo->eventName, "Command") == 0)
    {
        LOG_VERBOSE("- Command name was => %s\r\n", callbackInfo->tag);
    }
}

void sendToAzure(double value)
{
}

void count_pulse()
{
    pulse++;
}

void setup()
{
    Serial.begin(9600);

    connect_wifi(WIFI_SSID, WIFI_PASSWORD);
    connect_client(SCOPE_ID, DEVICE_ID, DEVICE_KEY);

    pinMode(pin, INPUT_PULLUP);
    attachInterrupt(digitalPinToInterrupt(pin), count_pulse, RISING);
}

void loop()
{
    if (isConnected)
    {
        pulse = 0;
        interrupts();
        delay(interval);
        noInterrupts();

        Serial.print("Pulses: ");
        Serial.print(pulse);
        Serial.print(", kWh per 30s: ");
        Serial.println(double(pulse) / double(pulses_per_kwh));

        char msg[64] = {0};
        int pos = 0, errorCode = 0;

        pos = snprintf(msg, sizeof(msg) - 1, "{\"energy\": %f}",
                       double(pulse) / double(pulses_per_kwh));
        errorCode = iotc_send_telemetry(context, msg, pos);
        msg[pos] = 0;

        if (errorCode != 0)
        {
            LOG_ERROR("Sending message has failed with error code %d", errorCode);
        }
        iotc_do_work(context);
    }
    else
    {
        iotc_free_context(context);
        context = NULL;
        connect_client(SCOPE_ID, DEVICE_ID, DEVICE_KEY);
    }
}
```

```bash
  - Is connected ? YES (64)
Pulses: 0, kWh per 30s: 0.00
  - - [MessageSent] event was received. Payload => {"energy": 0.000000}

  - Is connected ? NO (128)
  - ERROR: Client was not connected.
  - - iotc.dps : getting auth...
  - - iotc.dps : getting operation id...
  - - iotc.dps : getting host name...
```

obastemur, 2019-05-07:
Curious if it's an `snprintf` bug with `%f`. Could you try another value with `%d` ?

could you also please put `msg[pos] = 0;` before the send_telemetry?

anderl80, 2019-05-07:
Seems, I cannot use Serial COM, following works

```c++
void loop()
{
    if (isConnected)
    {
        unsigned long ms = millis();
        if (ms - lastTick > interval)
        {
            char msg[64] = {0};
            int pos = 0, errorCode = 0;

            lastTick = ms;
            pos = snprintf(msg, sizeof(msg) - 1, "{\"energy\": %f}",
                           double(pulse) / double(pulses_per_kwh));
            errorCode = iotc_send_telemetry(context, msg, pos);

            msg[pos] = 0;
            pulse = 0;

            if (errorCode != 0)
            {
                LOG_ERROR("Sending message has failed with error code %d", errorCode);
            }
        }

        iotc_do_work(context);
    }
    else
    {
        iotc_free_context(context);
        context = NULL;
        connect_client(SCOPE_ID, DEVICE_ID, DEVICE_KEY);
    }
}
```

obastemur, 2019-05-07:
make sense. stick with `LOG_VERBOSE`. it's a `serial.printf`

closed 2019-05-07 by obastemur