Azure/iot-central-firmware #78: c client for l476RG iothub

dianahinz, 2019-05-09:
I've been struggling to make azure iot c sdk work with my l476rg. No luck. Found iotc client here and confirmed that it fits into my l476RG with 128kb.

I played the sample application scenario and tested with azure iot central. What I'm supposed to do for iothub only? Is iotc expected to work with azure iothub only?

obastemur, 2019-05-10:
@dianahinz Have you shared the problem details to Azure IoT SDK C repository?

short answer: consider iotc client as a thin layer on top of platform soft layers. It supports `connection string` authentication (default hub authentication) as well as others. You shouldn't have any issue with connecting / using Azure Iothub. Because it's a thin layer, I encourage you to look at the API and see if you find everything you need. Azure IoT SDK C has full set of features.

Azure IoT Central is the combination of Azure IoT Device Provisioning Service and Azure IoT Hub services. Also comes with a builtin backend logic that is designed to scale and add additional value to your data. So, iotc embedded will put more info on your hub then you actually delivered. This is good for some and bad for others.

cumakeen, 2019-05-11:
I'm in the same boat. We made iotc client to work with our samd device.

Do you accept contributions to iotc? Where exactly it is hosted?

rmrewak, 2019-05-12:
Same as @cumakeen . Silently patching our copy of iotc. Any plan to update mbed sample? (#79 )

Thanks for the samples and good work but don't expect us to scan all microsoft documents and follow up with the developments by ourselves.