Azure/iot-central-firmware #81: AZ3166 Access Point Can't Connect To Network (MXCHIP)

stefanzvonar, 2019-06-04:
Using latest firmware (v2.1.1) I can't for the life of me get a connection (using many other wifi devices) to the MXCHIP's Access Point, to even be able to configure it.

Keeps saying "Can't connect to this network"  or "Access Denied".

Prompt on the chip's display is: 
Connect HotSpot:
AZ3166_SRSDFT
go-> 192.1638.0.1
PIN CODE XXXX

obastemur, 2019-06-04:
[mxchip_setup.html.zip](https://github.com/Azure/iot-central-firmware/files/3254906/mxchip_setup.html.zip)

You may use the html app from the zip file above and create a device binary with the necessary info instead. (your wifi ssid / password and app details). It's an offline http app.

obastemur, 2019-06-04:
I'm not sure why your device is not able to connect MXCHIP wifi endpoint though. Might be related to supported radio fq?

stefanzvonar, 2019-06-05:
Thank you @obastemur, you're a life saver, that offline .bin generator was exactly what I needed! :)

closed 2019-06-05 by obastemur