Azure/iot-central-firmware #83: HTTP Bash: Unterminated string. Expected delimiter.

ppoektos, 2019-07-02:
While using provided bash script I'm getting this error:
`undefined:1
{"errorCode":400012,"trackingId":"fa7462e6-0a2f-4316-bc69-56272819ba02","message":"Unterminated string. Expected delimiter: ". Path 'registrationId', line 1, position 25.","timestampUtc":"2019-07-02T14:18:27.8155458Z"}
                                                                                                                             ^

SyntaxError: Unexpected token . in JSON at position 125
    at JSON.parse (<anonymous>)
    at [eval]:1:8
    at Script.runInThisContext (vm.js:122:20)
    at Object.runInThisContext (vm.js:329:38)
    at Object.<anonymous> ([eval]-wrapper:6:22)
    at Module._compile (internal/modules/cjs/loader.js:776:30)
    at evalScript (internal/bootstrap/node.js:589:27)
    at startup (internal/bootstrap/node.js:265:9)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:622:3)`

This is code that I exactly have:
`OUT=$(curl -s \
  -H "authorization: ${AUTH}&skn=registration" \
  -H 'content-type: application/json; charset=utf-8' \
  -H 'content-length: 25' \
  -X PUT -d "{\"registrationId\":\"$DEVICE_ID\"}" \
  "https://global.azure-devices-provisioning.net/$SCOPEID/registrations/$DEVICE_ID/register?api-version=2018-11-01")`

What's wrong with **registrationId** here?

obastemur, 2019-07-02:
Please mind the `content-length: 25` . Any changes to device id etc. length must be represented in that number.

kartben, 2019-07-02:
Right. Actually, the content-length header is not required as curl automatically sets "Content-Length: 0" header when -d/--data flag is used. I will submit a PR.

closed 2019-07-02 by obastemur

ppoektos, 2019-07-03:
Hi.
After removing that content-length header I'm getting this:

```
Authenticating..
OK

SENDING =>  {"temperature":"34","voltage":"199","runtime":"18600","status":"OL CHRG"}
DONE

{ errorCode: 400004,
  trackingId: '20ed682d-fce1-43bc-bab2-7ee8c1a78979',
  message: 'Invalid idScope length',
  timestampUtc: '2019-07-03T07:21:20.5953793Z' }
```