Azure/iot-central-firmware #86: Corrupted Heap

bhjolly, 2019-07-29:
I just followed all the instructions for the ESP32 example and got everything set up, make flash completed successfully, make monitor looks like it works for a few seconds with successful connect and transmit but the ESP32 resets on what looks like a transmit confirmation:

`I (3755) event: sta ip: 192.168.1.114, mask: 255.255.255.0, gw: 192.168.1.1
I (3755) azure: Connected to AP success!
Initializing SNTP
ESP platform sntp inited!
I (3755) platform: The current date/time in Shanghai is: Mon Jul 29 18:10:13 2019
  - - IOTC: Registration status: CONNECTED
  - - IOTC: Registration status: ASSIGNING
  - - IOTC: Registration status: ASSIGNING
  - ConnectionString: HostName=x.azure-devices.net;DeviceId=x;SharedAccessKey=x
  - Is connected ? YES (64)
  - - [ConnectionStatus] event was received. Payload => None
  - SENT => {"T": 18}
  - - [MessageSent] event was received. Payload => {"T": 18}-�?-�?
CORRUPT HEAP: multi_heap.c:432 detected at 0x3ffd36c4
abort() was called at PC 0x4008990f on core 0
0x4008990f: multi_heap_assert at /home/ben/esp/esp-idf/components/heap/multi_heap.c:380
`
The only thing I changed was 'temp' to 'T' in the message payload but even after reverting I get the same result

albahmed23, 2019-08-08:
I have the exact same issue as well. Followed the instructions for ESP32 and get the multi_heap.c error. The CORRUPT HEAP assert.

It might have to do with the timestamp since there are some digits that are coming in after the payload that might cause the heap to overflow? 

Any ideas why this might be happening? Have you tried something new. 

Thank you!

bhjolly, 2019-08-11:
No idea sorry, I've bailed to AWS IoT Core for the moment (device code libraries seem more reliable)

obastemur, 2019-08-12:
/cc @Azure/azure-iot-sdk-c-admins 

@zhaodong2013062 You had issue with ESP32 too right?

How about https://github.com/espressif/esp-azure ?

bhjolly, 2019-08-12:
FYI that's for IoT Hub (requires a 'connection string'), not IoT Central (gives you 'scope', 'id', and 'key'). From memory I had some success with that when running a Hub but if you want Central then you need something else I think.