Azure/iot-central-firmware #89: pypi.org/project/iotc/ micropython callback incomplete - any plans to complete?

jaybode, 2019-08-30:
Hi All,

I notice that the Micropython version that uses umqtt.simple client has an empty callback function.

I finally got iotc to import running on esp32 with no errors (needed expanded PSRAM version)

This _mqttcb function is an empty for callback, so is doing nothing when calling a command initiated from IOTC portal; but received MQTT command when debugging with print.

`  def _mqttcb(self, topic, msg):
    #print('### topic: ' + str(topic))
    #print('### msg: ' + str(msg))
    #pass`

Is there any plans to complete library for Micropython version?  
ESP32 will obviously not work with paho-mqtt, and can see you have tried to support micropython umqtt.simple - but is incomplete on callback

I actually tried to complete the function based on your paho-mqtt working code, but I was having issues with the class and OOB (I am not strong on Pythton)
It worked on first sending a command to device from IOTC port, but second call would crash the micropython loop.

Any help/pointer appreciated. 

I am a hobby programmer, and getting ESP32 working with IOTC would be wonderful. This is since IOTC commands supported Microsoft flow, so once working its easy to build a PowerApps mobile application to send commands to a registered device securely - and within an enterprise O365 customers operating environment and authentication/identity.

I want to showcase a video at the end how easy any O365 customer can build/publish a mobile app to control an IOT device that is registered with IOTC.

Thanks,

Jason

Cheers,

Jason

obastemur, 2019-08-30:
It should be feature complete with micropython. There must be something else going on. Which version of the mp?

jaybode, 2019-08-30:
I was using iotc-0.3.3.tar.gz from PiPy a few weeks back.  I see a new version now 0.3.4

I see 0.3.4 it is the same - but I didnt test yet. unless I am missing something.

Line 135 sets the callback when micropython detected:
`__self._mqtts.set_callback(__self._mqttcb)`

The callback is empty.
`
  def _mqttcb(self, topic, msg):
    # NOOP
    pass
`

State changes seem to work when changing a state value on IOTC portal.

But when sending a command from IOT portal (or through Microsoft flow) its received to device OK, but does nothing. It does log out OK when adding print statements.

`def _mqttcb(self, topic, msg): `
  `  #print('### topic: ' + str(topic))` 
  `  #print('### msg: ' + str(msg)) `

paho-mqtt version that is called on non-micro-python system: _self._onMessage  works fine.

I tired copying/editing the paho-mqtt _onMessage equivalent function over into _mqttcb function but had no luck e.g.
line 479: 
`  def _onMessage(self, client, _, data):`

It would great if someone would test micropython version and complete working example.

obastemur, 2019-08-30:
`def _mqttcb(self, topic, msg): # NOOP pass` is by design. Which MP (micropython) version?

jaybode, 2019-08-30:
This was my attempt to complete the function so commands work on micropython.
It receives/prints command once. Second time crashes loop.

` def _mqttcb(self, topic, msg):

    #print('### topic: ' + str(topic))
    #print('### msg: ' + str(msg))
    #pass

    if topic == None:
      LOG_IOTC("WARNING: (_onMessage) topic is None.")
      return

    if msg != None:
      try:
        msg = msg.decode("utf-8")
      except:
        msg = str(msg, 'utf-8')

    if topic != None:
      try:
        topic = topic.decode("utf-8")
      except:
        topic = str(topic, 'utf-8')

    LOG_IOTC("- iotc :: _onMessage :: topic(" + topic + ") msg(" + msg + ")", IOTLogLevel.IOTC_LOGGING_ALL)

    if topic.startswith('$iothub/'): # twin
      # DO NOT need to echo twin response since IOTC api takes care of the desired messages internally
      # if topic.startswith('$iothub/twin/res/'): # twin response
      #   self._handleTwin(topic, msg)
      #
      if topic.startswith('$iothub/twin/PATCH/properties/desired/') or topic.startswith('$iothub/twin/res/200/?$rid='): # twin desired property change
        self._echoDesired(msg, topic)
      elif topic.startswith('$iothub/methods'): # C2D
        index = topic.find("$rid=")
        method_id = 1
        method_name = "None"
        if index == -1:
          LOG_IOTC("ERROR: C2D doesn't include topic id")
        else:
          method_id = topic[index + 5:]
          topic_template = "$iothub/methods/POST/"
          len_temp = len(topic_template)
          method_name = topic[len_temp:topic.find("/", len_temp + 1)]
        
        # JASE -troubleshoot here
        ret = MAKE_CALLBACK(self, "Command", msg, method_name, 0)
        ret_code = 200
        ret_message = "{}"
        if ret.getResponseCode() != None:
          ret_code = ret.getResponseCode()
        if ret.getResponseMessage() != None:
          ret_message = ret.getResponseMessage()

        next_topic = '$iothub/methods/res/{}/?$rid={}'.format(ret_code, method_id)
        LOG_IOTC("C2D: => " + next_topic + " with data " + ret_message  + " and name => " + method_name, IOTLogLevel.IOTC_LOGGING_ALL)
        (result, msg_id) = self._mqtts.publish(next_topic, ret_message, qos=0)
        if result != MQTT_SUCCESS:
          LOG_IOTC("ERROR: (send method callback) failed to send. MQTT client return value: " + str(result))
      else:
        if not topic.startswith('$iothub/twin/res/'): # not twin response
          LOG_IOTC('ERROR: unknown twin! {} - {}'.format(topic, msg))
    else:
      LOG_IOTC('ERROR: (unknown message) {} - {}'.format(topic, msg))`

Main program follows example from the provided sample code:

import iotc
from iotc import IOTConnectType, IOTLogLevel
from random import randint

iotc = iotc.Device(scopeId, deviceKey, deviceId, IOTConnectType.IOTC_CONNECT_SYMM_KEY)
iotc.setLogLevel(IOTLogLevel.IOTC_LOGGING_ALL) # IOTC_LOGGING_API_ONLY

gCanSend = False
gCounter = 0

def onconnect(info):
  global gCanSend
  print("- [onconnect] => status:" + str(info.getStatusCode()))
  if info.getStatusCode() == 0:
     if iotc.isConnected():
       gCanSend = True

def onmessagesent(info):
  print("\t- [onmessagesent] => " + str(info.getPayload()))

def oncommand(info):
  print("command name:", info.getTag())
  print("command value: ", info.getPayload())

def onsettingsupdated(info):
  print("setting name:", info.getTag())
  print("setting value: ", info.getPayload())

iotc.on("ConnectionStatus", onconnect)
iotc.on("MessageSent", onmessagesent)
iotc.on("Command", oncommand)
iotc.on("SettingsUpdated", onsettingsupdated)

iotc.connect()

while iotc.isConnected():
  iotc.doNext() # do the async work needed to be done for MQTT
  if gCanSend == True:
    if gCounter % 20 == 0:
      gCounter = 0
      print("Sending telemetry..")
      iotc.sendTelemetry("{\"temp\": " + str(randint(20, 45)) + "}")

    gCounter += 1

jaybode, 2019-08-30:
I see your new question:

I am using a fork to use SPIRAM. Otherwise I cannot import iotc library without running out of memory.

https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo

Good question on version of Micro-python. I am not sure, I need to check.

But if the NOOP section by design, then the device does nothing when a command is sent to it when running micropython??

I see the other function for when paho-mqtt is imported instead, it properly logs out response and ignores twin response etc. when a command is sent to the device.

Much appreciated the dialog.

jaybode, 2019-08-30:
I can load Micropython on Linux version and test and report back results, but confident I will get the same results when it detects usimple.mqtt client and used the different function for callback - which to me is empty compared to if import on normal python and other function is called.

jaybode, 2019-08-31:
I figured out the bug in my code in filling in the missing function _mqttcb - and all working now for me. The missing function contents cannot be by design sorry. It simply doesn't work without this function when running in micropython and umqtt.simple

I think you should discover yourself later that when running in Micropython. The 'settings update' and 'commands' when run from IOTC portal does nothing on the device with debug on - and hangs/timeout waiting for response on ITOC side. This is because _mqttcb is used via __self._mqtts.set_callback. 

Running normal python it works fine since it will use __self._onMessage and the function _onMessage instead - which is complete.

Thanks for the fast response anyway on trying to help. Fantastic work all this.

Here is my working code for anyone finds this post, and having same issue running from micro-python:
`
def _mqttcb(self, topic, msg):

    if topic == None:
      LOG_IOTC("WARNING: (_onMessage) topic is None.")
      return

    if msg != None:
      try:
        msg = msg.decode("utf-8")
      except:
        msg = str(msg, 'utf-8')

    if topic != None:
      try:
        topic = topic.decode("utf-8")
      except:
        topic = str(topic, 'utf-8')

    LOG_IOTC("- iotc :: _onMessage :: topic(" + topic + ") msg(" + msg + ")", IOTLogLevel.IOTC_LOGGING_ALL)

    if topic.startswith('$iothub/'): # twin

      if topic.startswith('$iothub/twin/PATCH/properties/desired/') or topic.startswith('$iothub/twin/res/200/?$rid='): # twin desired property change
        self._echoDesired(msg, topic)
      elif topic.startswith('$iothub/methods'): # C2D
        index = topic.find("$rid=")
        method_id = 1
        method_name = "None"
        if index == -1:
          LOG_IOTC("ERROR: C2D doesn't include topic id")
        else:
          method_id = topic[index + 5:]
          topic_template = "$iothub/methods/POST/"
          len_temp = len(topic_template)
          method_name = topic[len_temp:topic.find("/", len_temp + 1)]
        
        ret = MAKE_CALLBACK(self, "Command", msg, method_name, 0)
        ret_code = 200
        ret_message = "{}"
        if ret.getResponseCode() != None:
          ret_code = ret.getResponseCode()
        if ret.getResponseMessage() != None:
          ret_message = ret.getResponseMessage()

        next_topic = '$iothub/methods/res/{}/?$rid={}'.format(ret_code, method_id)
        LOG_IOTC("C2D: => " + next_topic + " with data " + ret_message  + " and name => " + method_name, IOTLogLevel.IOTC_LOGGING_ALL)
        # (result, msg_id) = self._mqtts.publish(next_topic, ret_message, qos=0) # ORG
        self._mqtts.publish(next_topic, ret_message)
        msg_id = 0
        self._messages[str(msg_id)] = None
      else:
        if not topic.startswith('$iothub/twin/res/'): # not twin response
          LOG_IOTC('ERROR: unknown twin! {} - {}'.format(topic, msg))
    else:
      LOG_IOTC('ERROR: (unknown message) {} - {}'.format(topic, msg))

`

closed 2019-08-31 by jaybode