Azure/iot-central-firmware #94: Getting the template details

edsase, 2019-10-24:
Hi, 
It will be great to be able to get the template details from IOTC to the device using the API. Use case is: once the template has been changed, the device code can get notified or be checking regularly and only sends the telemetry indicated on the template to the cloud. At the moment, if the template changes, like when new telemetry variables are added to a template, the programmer would have to adjust the code on the device manually. The function I'm requesting for will help to automate this process. Thanks.

kartben, 2019-10-28:
Hi @edsase -- I would suggest you have a look at IoT Plug-and-Play, and try out the public preview of Plug and Play on IoT Central. It's meant to help the server (in your case IoT Central) essentially know in real-time what capabilities a device has in real-time, i.e know what telemetry it sends etc.

edsase, 2019-10-28:
Thanks for your suggestion @kartben. What I am looking for though is a function that would be called from the device via the api that gets the template details from the server unto the IOT device. I saw IOTC Central has a get_properties() function. A similar function, call it get_template_details() would be really handy.