Azure/iot-central-firmware #95: IoTC python client connect  exception: InvalidURL

georgegoglodze, 2019-11-01:
Hi everyone, 
I am using python client to push data to IOTC. https://pypi.org/project/iotc/
Everything was working fine but just discovered that all of my iotc python clients are experiencing the exception listed below. Please advise if you know workaround.

```
Traceback (most recent call last):
  File "2.py", line 43, in <module>
    iotc.connect()
  File "/usr/local/lib/python2.7/dist-packages/iotc/__init__.py", line 635, in connect
    content = _request(self, target.geturl(), "PUT", body, headers)
  File "/usr/local/lib/python2.7/dist-packages/iotc/__init__.py", line 277, in _request
    return _doRequest(device, target_url, method, body, headers)
  File "/usr/local/lib/python2.7/dist-packages/iotc/__init__.py", line 269, in _doRequest
    conn.request(method, socket_buffer)
  File "/usr/lib/python2.7/httplib.py", line 1073, in request
    self._send_request(method, url, body, headers)
  File "/usr/lib/python2.7/httplib.py", line 1107, in _send_request
    self.putrequest(method, url, **skips)
  File "/usr/lib/python2.7/httplib.py", line 960, in putrequest
    % (url, match.group()))
httplib.InvalidURL: URL can't contain control characters. 'https://global.azure-devices-provisioning.net/0ne0008C6C6/registrations/smart1/register?api-version=2018-11-01 HTTP/1.1\r\nHost: global.azure-devices-provisioning.net\r\ncontent-type: application/json; charset=utf-8\r\nuser-agent: iot-central-client/1.0\r\naccept: */*\r\naccept-encoding: gzip, deflate\r\nauthorization: SharedAccessSignature sr=0ne0008C6C6%2Fregistrations%2Fsmart1&sig=MbFq%2F0Cgtx70EIInc7P1fE7o46NjoIcSTeZeieufzjY%3D&se=1572506101&skn=registration\r\nContent-Length: 27\r\n\r\n{"registrationId":"smart1"}' (found at least ' ')
root@iotdev:~/SmartTable# python 2.py
Traceback (most recent call last):

```

edsase, 2019-11-04:
Oh, I'm getting the same error. In my case, it only happens when I try to run the code in a docker container. It runs well on my laptop and on a RaspberryPi host but fails with the above error in the docker container  as I explained [here](https://stackoverflow.com/questions/58679047/invalid-url-error-is-raised-when-running-azure-iotc-connect-from-within-a-pyth). Will indeed appreciate a workaround.

kartben, 2019-11-05:
Could you confirm which version of the iotc pip package you are running?
Doesn't look like it's the latest version, from the stack trace? I am not
sure why your URL would be this giant string with not only the URL but also
HTTP header and payload..? so would be great to know/confirm what version
of the lib you are using.

edsase, 2019-11-05:
IOTC version on both laptop and raspberryPi is 0.3.4 which is the latest version. Like I mentioned above, it works well for me both on my Windows 10 laptop and RaspberryPi host. I only get this error when the code is ran in a Docker container within a RaspberryPi host.

georgegoglodze, 2019-11-05:
I am using the latest version of IOTC as well and I receive the error
message on both Docker and Ubuntu server.

kartben, 2019-11-05:
Can you check that your iotc __init.py__ matches https://github.com/obastemur/iot_client/blob/master/python/src/iotc/__init__.py? The line numbers you have in your stack trace don't line up with 0.3.4 so this is definitely weird...

@obastemur or @firedog1024 might have some thoughts?

obastemur, 2019-11-05:
Were you able to ping the azure iot global endpoint from docker? DNS resolver configuration might be the issue

obastemur, 2019-11-05:
Alternatively, I would look at the encoding of the file.

edsase, 2019-11-05:
I had pip 18.6 and updated to pip-19.3.1 (latest version). I rebuilt the image (after clearing old layers) but still got the same result. 

@obastemur, I didn't set up any network in the container but used only default. Which encoding are you referring to? I am not doing any encoding but im guess is being done by the IOTC package? Again, the really weird thing here is the script only fails in the container but not in the host.

kartben, 2019-11-05:
What really looks weird to me is that apparently it's trying to connect to an URL that has headers and payload and stuff, all as part of the URL

```
'https://global.azure-devices-provisioning.net/0ne0008C6C6/registrations/smart1/**register?api-version=2018-11-01 HTTP/1.1\r\nHost: global.azure-devices-provisioning.net\r\ncontent-type: application/json; charset=utf-8\r\nuser-agent: iot-central-client/1.0\r\naccept: */*\r\naccept-encoding: gzip, deflate\r\nauthorization: SharedAccessSignature sr=0ne0008C6C6%2Fregistrations%2Fsmart1&sig=MbFq%2F0Cgtx70EIInc7P1fE7o46NjoIcSTeZeieufzjY%3D&se=1572506101&skn=registration\r\nContent-Length:
```

Hence the error message about improper character ' '

obastemur, 2019-11-05:
That’s why I asked about the encoding or any tampering to file that may brake cr+lf

lucadruda, 2019-11-06:
version 0.3.4 works well when running on python 2.7 and gives no error when accessing the URL.

georgegoglodze, 2019-11-06:
@lucadruda Can you please share you python client? 
As for me I tested python2.7 and 3.6 on Ubuntu 18 and Centos 7 but still see the same URL error

lucadruda, 2019-11-06:
It works for me on 2.7 with same client (0.3.4).
Anyway PR with fix for both 2.7+ and 3+ is open and waiting for code review.

lucadruda, 2019-11-07:
Fix is available. Please pull latest version (0.3.5) available on [https://pypi.org/project/iotc/](https://pypi.org/project/iotc/)
Closing the issue

closed 2019-11-07 by lucadruda

georgegoglodze, 2019-11-07:
Worked like a charm! Thanks everyone.

edsase, 2019-11-07:
Yes! Worked perfectly! Thanks Guys