Azure/iot-central-firmware #96: Encrypted telemetry data

abdjiber, 2019-11-08:
Hi,

I am storing my real device data connected to iot central with a blob storage. I would like to apply AI on the data in Azure ML studio however the telemetry data are encryted. How could I decrypt them ? Or avoid the encryption ? Here is an example of what I got in the Body on my JSON's files:

"Body":"eyAJCQkJImFpcl9odW1pZGl0eSI6IDEyLCAJCQkJImFpcl90ZW1wZXJhdHVyZSI6IDMsIAkJCQkicHJlc3N1cmUiOiA1LCAJCQkJInNvaWxfaHVtaWRpdHkiOiAzLCAJCQkJIndpbmRfZGlyZWN0aW9uIjogMjEsIAkJCQkid2luZF9zcGVlZCI6IDQxLCAJCQkJImlycmlnYXRpb24iOiAwLCAJCQkJIndpbmRfZ3Vlc3QiOiAxNTksIAkJCQl9" 

I noticed that the data from a simulated device are not encrypted. Could anyone help me please it's very urgent ? I search for a long time on the web and couldn't find a solution. I saw in [this](https://docs.microsoft.com/en-us/azure/iot-central/core/concepts-connectivity ) documentation in Security section that data from real devices are encrypted but how to decrypt them ?

kartben, 2019-11-09:
You need to set the content type of the messages you sent to IoT Central to
"application/json" and the encoding to "utf-8"

abdjiber, 2019-11-09:
Thanks for your response. I am using the iotc python package. If I set the systemProperties = {"content-type":"application/json", "content-encoding":"utf-8"} after one message is sent the connection status become 1 and the telemetry sending stop.

aimjarozas, 2020-02-06:
Hi, u have the same problem, this have some solutions?:
iotc.sendTelemetry("{\"temp\": " 20"},{"contentType":"application/json","contentEncoding":"utf-8"}): after one message is sent the connection status become 1 and the telemetry sending stop.

If I put the follow:
iotc.sendTelemetry("{\"temp\": " 20"},{"contentEncoding":"utf-8"})

The system put the contentEncoding in Properties, but this must to appear in "systemProperties:
{"EnqueuedTimeUtc":"2020-02-05T11:25:14.6600000Z","Properties":{"contentEncoding":"UTF-8"},"SystemProperties":{"connectionDeviceId":"c08bccb2-f98b-4970-8c23-64f4cad04778","connectionAuthMethod":"{\"scope\":\"device\",\"type\":\"sas\",\"issuer\":\"iothub\",\"acceptingIpFilterRule

Do you have some solutions?