Azure/iot-central-firmware #100: mqtt-log : failed to receive on socket: [WinError 10054] An existing connection was forcibly closed by the remote host

dmitriyteteruk, 2019-11-24:
Hi! I am getting error that result in Python app stop to send a message. Python 3.8.0, iotc 0.3.5, paho-mqtt from iotc 1.5.0, httplib2 from iotc 0.14.0. Any suggestions/ideas?
```
1574498785.245122 mqtt-log : failed to receive on socket: [WinError 10054] An existing connection was forcibly closed by the remote host
1574498785.2461307 - iotc :: _onDisconnect :: rc = 1
1574498785.2461307 - iotc :: MAKE_CALLBACK :: ConnectionStatus
- [onconnect] => status:1
```

kobidetrenewan, 2019-11-24:
I saw it too in my code. Once it is disconnected, it never recovers.

lmasieri, 2019-12-12:
Hi @kobidetrenewan @dmitriyteteruk  if you're still experiencing the disconnect issue, can you please DM me your app ID  (can be found in Admin section), the device ID, and the SDK language you're using. We're looking to troubleshoot this.

kobidetrenewan, 2019-12-12:
Hi Luis,
I found a mistake in my logic:
I expected the on_connect() event to be re-triggered once the connection is reestablished and there I modified a private "canSend" variable, which dictates if the central loop tries to send the device data to the IOTC.. 
But the on_connect() event never re-triggered again.
Now I changed the test of canSend to a call to the function iotc.isConnected() instead, and the communication is recovered properly.

Thank you,
Kobi

lmasieri, 2019-12-13:
Thanks for confirming @kobidetrenewan

closed 2020-03-16 by dmitriyteteruk