Azure/iot-central-firmware #103: Error parsing device message from IoT Central on MXChip

colindembovsky, 2019-11-27:
I have compiled and uploaded the MXChip Avanced sample. I am able to register and receive telemetry from the board. However, the commands do no not work.

If I enter "5" into the CountDown command from IoT Central, the message gets to the MXChip but there appears to be an error parsing the packet:

```
Error at c:\users\colin\appdata\local\temp\arduino_build_354959\sketch\src\iotc\common\json.h:23
	JSON value is not an object
X - Error at c:\users\colin\appdata\local\temp\arduino_build_354959\sketch\src\iotc\common\json.h:39
	json data: 5
```

The same behavior is displayed for all of the other commands.

radik86, 2020-08-27:
I have similar issue.
Have you solved it?

radik86, 2020-08-31:
I had similar Issue. My issue was that the method json.getStringByName("displayedValue"); in registeredMethodHandlers.cpp was not able to find the property "displayedValue" and hence caused an null exception. By skipping this part and using the payload instead of the json.getStringByName("displayedValue"); I was able to achieve my target.

Just comment this part:

const char * text = json.getStringByName("displayedValue");
if (text == NULL) {
LOG_ERROR("Object doesn't have a member 'displayedValue' : %s", text);
return 0;
}