Azure/iot-central-firmware #107: Python - how to catch cloud to device error messages?

Frostlock, 2020-01-08:
I'm using the example code from pypi
When sending properties I sometimes mess up and the JSON is not accepted by the IOT hub.
With full logging enabled this shows up in the console output but I would like to catch this in a cleaner way. It would be nice to have a "MessageReceived" to subscribe to.
Or is there a proper way of doing this which I currently don't see :)

To reproduce add the following to the python example:
`iotc.sendProperty('{"broken": json}')`

With full logging enabled this shows the following error message in the console:
```
1578493282.9805958 mqtt-log : Received PUBLISH (d0, q0, r0, m0), '$iothub/twin/res/200/?$rid=0', ...  (119 bytes)
1578493282.9805958 - iotc :: _onMessage :: topic($iothub/twin/res/200/?$rid=0) payload(b'{"desired":{"$version":1},"reported":{"totalStorage":{"value":157},"guiVersion":{"value":"Summer 2020"},"$version":12}}')
1578493282.9805958 - iotc :: _echoDesired :: $iothub/twin/res/200/?$rid=0
1578493283.1068795 mqtt-log : Received PUBLISH (d0, q0, r0, m0), '$iothub/twin/res/400/?$rid=1578493282', ...  (226 bytes)
1578493283.1068795 - iotc :: _onMessage :: topic($iothub/twin/res/400/?$rid=1578493282) payload(b'{"errorCode":400004,"trackingId":"408808d6126145bba779c843dbfe97c0-G:9-TimeStamp:01/08/2020 14:21:23","message":"Error occured while parsing message payload as UTF-8 JSON object.","timestampUtc":"2020-01-08T14:21:23.0283507Z"}')

```

dmitriyteteruk, 2020-03-13:
[Check format of JSON](https://jsonformatter.curiousconcept.com/) output - it is incorrect.
[image]

Frostlock, 2020-03-13:
Thanks for the feedback :) I see my question might have been confusing. The key point is not the JSON formatting. The key point is how to catch errors that the IOT central platform sends back. 
So to be specific: How can I access the 
`"message":"Error occured while parsing message payload as UTF-8 JSON object."`
part of the log above programmatically? If the IOT central sends an error message back I want to be able to know this in my program. I hope that helps to clarify my question :)