Azure/iot-central-firmware #108: Python - Geopoint property fails to upload

Frostlock, 2020-01-13:
I'm using the IOTC firmware (through python). I have been able to upload location data as telemetry. However I have not been able to upload location data as a basic property. 
This should be possible, as illustrated by one of the answers on my [question on stackexchange](https://stackoverflow.com/questions/59661241/using-a-geopoint-property-in-azure-iot-central)
but it doesn't work for me using Python through the Azure IOT Central firmware.

Since my device isn't supposed to move around I wasn't planning on sending location as telemetry. I would prefer just to send the location once as a device property when the device boots and connects to Azure IOT Central. 

I'm formatting my JSON like this
`{"location":{"lon":3.837585, "lat":41.174130}}`
and I have also tried this
`{"location":{"value": {"lon":3.837585, "lat":41.174130}}}`

Am I doing something wrong or is the python firmware not handling this properly?

thanks!

prasannapattam, 2020-02-19:
This worked for me
`{"location":{"lat":48.9332077,"lon":8.9611761}}`

prasannapattam, 2020-02-20:
I also have same problem. I was able to get the location by telemetry. But not as property.

I couldn't get any Property to work. I defined a writable string property in the Template and using the following C# code to send data and this is not working.
`message.Properties.Add("currentloc", DateTime.Now.ToLongTimeString())`

Is there any precautions to define a property in Templates?

dmitriyteteruk, 2020-03-13:
Use below JSON format of geo location properties or telemetry - works for me.
{
  "Location":{
    "value":{
      "lat":55.752078,
      "lon":37.619435
    }
  }
}

Frostlock, 2020-03-16:
Thanks @dmitriyteteruk 
I have pulled up my code from before. No changes made on client side, no changes made on my device template. Both the telemetry and the property upload of location data now work correctly.
I'm wondering if something got fixed in the backend :)
If anyone knows where I can find the Azure IOT Central release notes please share.

Closing this ticket since it looks to be solved.

closed 2020-03-16 by Frostlock

dmitriyteteruk, 2020-03-16:
@Frostlock , it was working in legacy application templates around 1,5 years and it start to work for single device in current app template some time ago (1-2 month) while it is still marked as Know Issue that is not working.

Frostlock, 2020-03-16:
@dmitriyteteruk interesting, where do you find that information? We are currently on the public preview version and it would be nice to have a look at the known issues.

dmitriyteteruk, 2020-03-16:
@Frostlock, Preview word is not appears anymore in ex-Preview (Current) Apps when you are creating it. [Official announcement](https://azure.microsoft.com/ru-ru/blog/location-and-maps-in-azure-iot-central-powered-by-azure-maps/) of Maps support was in June 2018. I found that Preview start to be current sometime ago, then Preview announcement stooped to appear in documentation. And I am testing current app with Geo for single device and it works.
Official Known issues [available here](https://docs.microsoft.com/bs-latn-ba/Azure/iot-central/core/overview-iot-central#known-issues)